PX4/jMAVSim #2: Provide keyboard shortcuts to switch the camera view

LorenzMeier, 2015-07-02:
@molecule Would it be possible to do the view switching here using keyboard shortcuts instead of hardcoding them? Can we print an instruction line on application startup about that?
https://github.com/PX4/jMAVSim/blob/master/src/me/drton/jmavsim/Simulator.java#L122-L140

molecule, 2015-07-07:
Yes I can add that. So it looks like the options are: FPV, on vehicle with gimbal, and static while pointing to vehicle, is that right? What about a flag -v (for view) and then either fpv, gimbal or static as the options? So the usage string could look like this: java -cp lib/*:out/productionjmavsim.jar me.drton.jmavsim.Simulator -udp <ip address>:14553 -v gimbal

Does that sound good to you?

LorenzMeier, 2015-07-07:
It sounds good, but keyboard hotkeys would be even better. I think you will often end up wanting to switch views as you fly.

DrTon, 2015-07-08:
Agree, good idea with hotkeys!

LorenzMeier, 2015-09-21:
Implemented on master, closing.

closed 2015-09-21 by LorenzMeier