PX4/jMAVSim #21: Java compilation on 1.8 broken

LorenzMeier, 2016-04-01:
@mpaperno I now get this when trying to run your updated version:

```
    [javac]         return String.format(ff + ff + ff, vec.getX(), vec.getY(), vec.getZ());
    [javac]                                                           ^
    [javac]   symbol:   method getY()
    [javac]   location: variable vec of type Vector3d
    [javac] /Users/lorenzmeier/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/ReportUtil.java:20: error: cannot find symbol
    [javac]         return String.format(ff + ff + ff, vec.getX(), vec.getY(), vec.getZ());
    [javac]                                                                       ^
    [javac]   symbol:   method getZ()
    [javac]   location: variable vec of type Vector3d
    [javac] 18 errors
```

LorenzMeier, 2016-04-01:
I fixed this by returning to the current standard API (.z, not .getZ()).

closed 2016-04-01 by LorenzMeier

mpaperno, 2016-04-01:
Lorenz, not sure what to tell you... I just tried a full rebuild here to make sure, and no problems.  And Travis seems to agree.  I'm using latest (I think) JDK on Win64 (1.8.0_74).  Also builds with 1.7.0_80.

OK just saw your update... I was about to suggest trying that.  Strange that you get the error though. Which Java version and OS?

-Max