PX4/jMAVSim #22: Crash on load due to missing image file

LorenzMeier, 2016-04-01:
@mpaperno Did you may forget to commit the compass image? I get a crash on boot because the image is not found:

```
Exception in thread "main" java.lang.IllegalArgumentException: input == null!
	at javax.imageio.ImageIO.read(ImageIO.java:1388)
	at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.<init>(Visualizer3D.java:707)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:114)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:209)
	at me.drton.jmavsim.Simulator.main(Simulator.java:611)
^C
```

mpaperno, 2016-04-01:
Lorenz, did you copy everything from `/environment` to `/out/production/environment` ?  Or wherever you're running it from.  If you run the `ant copy_res` it will copy everything for you.

-Max

LorenzMeier, 2016-04-01:
Yes, and that doesn't seem to fix it. I verified it did indeed copy. How do you exactly launch? I launch via:
```
cd jMAVSim
java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -udp 127.0.0.1:14560
```

LorenzMeier, 2016-04-01:
Using your launch instructions results in:
```
JavaVM WARNING: JAWT_GetAWT must be called after loading a JVM
AWT not found
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.NoClassDefFoundError: apple/awt/CGraphicsDevice
	at javax.media.j3d.GraphicsConfigTemplate3D.<clinit>(GraphicsConfigTemplate3D.java:55)
	at com.sun.j3d.utils.universe.SimpleUniverse.getPreferredConfiguration(SimpleUniverse.java:368)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:112)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:209)
	at me.drton.jmavsim.Simulator.main(Simulator.java:611)
	... 5 more
Caused by: java.lang.ClassNotFoundException: apple.awt.CGraphicsDevice
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 10 more
```

mpaperno, 2016-04-01:
I use 
```
ant create-run-jar copy_res
cd out/production
java -jar jmavsim_run.jar
```
But the location of all the resources hasn't changed... so if your launch command used to work, I don't see why it wouldn't now.

LorenzMeier, 2016-04-01:
A quick Google search suggests that the particular sub-version of Java3D you've now required will only work on Linux, as Oracle is incapable of making sure all other platforms are up to date.

I hope that I'm wrong and its something that can be fixed through proper configuration.

Would you considering trying to find a Mac OS sparring partner helping you to debug it? Otherwise we might have to revert or work around a few things, since the beauty of this simulator is its portability. We don't need yet another Gazebo-style thing that is not very portable, but has really good graphical abilities.

mpaperno, 2016-04-01:
Seems like running the jmavsim_run.jar version runs into the issue mentioned in the readme: https://github.com/PX4/jMAVSim#java-3d   Did you try adding `-Djava.ext.dirs=` to the call?

I didn't replace the version of java3d or change any requirements that I know of.  Or change the path to the resources. Does the old version of jMAVSim still run on your system?

I have a Mac VM I can try this on.  Won't be right away though.

-Max

mpaperno, 2016-04-01:
What Java version are you using, and which OS X exactly?

LorenzMeier, 2016-04-01:
You've nailed it, I missed that difference. Its booting now, but upsets the PX4 SITL (either initially sending garbage or because of a timing change). Can you rule out that the fields are initially not properly initialised and I should be looking for timing first?

LorenzMeier, 2016-04-01:
Fixed in d31a7a5.

LorenzMeier, 2016-04-01:
Bad data still persists. But I don't know yet if just a timing change or actual garbage sent.

closed 2016-04-01 by LorenzMeier

mpaperno, 2016-04-01:
I can't rule anything out, but I was very careful to keep all the defaults as they were. And any new settings should have defaults which don't affect operation.  I did make some improvements in performance, and on the other hand added more overhead (in the HUD overlay, adding gimbal model, etc.).  So if timing is very critical, that could be affected I suppose.  I run it at 800Hz, and our flight controller needs data at 400Hz (ideally)... it runs buttery smooth.

-Max

mpaperno, 2016-04-01:
The reporting panel is very useful for verifying what the sim "thinks" is going on.  Show all sensor outputs, etc.