PX4/jMAVSim #23: Still crash on load due to missing image file

muyangren499, 2016-04-04:
@mpaperno  @LorenzMeier
     I have a problem in anting the project jmavsim.  I also get a crash on boot because the image is not found:

Exception in thread "main" java.lang.IllegalArgumentException: input == null!
    at javax.imageio.ImageIO.read(ImageIO.java:1388)
    at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.<init>(Visualizer3D.java:707)
    at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:114)
    at me.drton.jmavsim.Simulator.<init>(Simulator.java:209)
    at me.drton.jmavsim.Simulator.main(Simulator.java:611)
I also used followed code, but it still report this errors. 
  java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator

In aternate build method. the terminal report  that the error is followed:

- BUILD FAILED
Target "creat_run_jar" does not exist in the project "jMAVSim". 

OS:Ubuntu 14.04 64bit
FIle:/home/user

mpaperno, 2016-04-04:
```
BUILD FAILED Target "creat_run_jar" does not exist in the project "jMAVSim". 
                         ^
```
Please check your spelling.

I don't know why the images don't work when trying to start it the "old way."  It shouldlook in `environment/` folder relative to the executing directory.  

Someone could try to debug the line throwing the error -- perhaps the way the image is being loaded (the classLoader) is causing the problem.  There are quite a few ways to load a URL/path in Java... one of them should work (as an example, there's another way used in `KinematicObject.modelFromFile()`).

```java
// load and scale compass image for overlay
Image img = ImageIO.read(getClass().getClassLoader().getResource(TEX_DIR + COMPASS_IMG));
```

-Max

muyangren499, 2016-04-04:
Thank you, Max. I will try it.  I just copied the code" ant create_run_jar copy_res" to the terminal.  I am a newer of Ubuntu,since the jMAVSIm is work in WIndows.

muyangren499, 2016-04-04:
@mpaperno Thank you very much for your work. the code is work after you fixed in my somputer

muyangren499, 2016-04-04:
in my computer

mpaperno, 2016-04-04:
Good to hear, and thanks for letting me know!

-Max

muyangren499, 2016-04-04:
@mpaperno  Great job! I will test controller  in the next period of time

PR #24 (2016-04-04): Fix overlay compass image loading error
Fixes #23 and #22 .
closed 2016-04-04 by LorenzMeier