PX4/jMAVSim #28: Pixhawk won't take off...

lovettchris, 2016-06-23:
I'm trying to do hardware in the loop simulation with Pixhawk and JMAVSim but take off command is doing nothing.  I can arm the pixhawk, and send take off commands, and Pixhawk is processing them (mc_att_control is generating thrust) but it seems the **actuator_controls** never turns into **actuator_outputs** which means it never shows up in mavlink HIL_CONTROLS message.  

Is there some additional setup I'm missing?  I've tried playing with pwm_out_sim with no luck... seems to me in HITL mode someone should intercept actuator_control and turn around and publish it as actuator_output, and pwm_out_sim seems to be doing that, but it does other stuff trying to drive the actual /dev/pwm_output0 which fails because that driver already exists. 

So is there some setup I'm missing here?  How to get px4io to pipe PX4IO_PAGE_CONTROLS right back in PX4IO_PAGE_SERVOS so that it gets picked up and published as ORB_ID(actuator_outputs) ?

PS : I want to do it with no motors actually attached to the Pixhawk of course.  Just bare Pixhawk on my desk.

dagar, 2016-06-23:
Did you select a HIL airframe in QGC (HIL Quadcopter X)?

lovettchris, 2016-06-23:
I don't see that option, (I see this http://www.lovettsoftware.com/LovettSoftware/scratch/qgc.png)
How do I get HIL Quadrocopter X installed?  And what does this model do exactly to make this work?

dagar, 2016-06-23:
Simulation is grouped together so the image displayed for the group is
actually a plane. It's 2 down from your current selection (row 3, column 3).

lovettchris, 2016-06-23:
Doh! Didn't see that, thanks.

closed 2016-06-23 by lovettchris