PX4/jMAVSim #29: Is HIL_STATE_QUATERNION missing in MAVLinkHILSystem.java ?

clovett, 2016-07-14:
The PX4 stack seems to expect HIL_STATE_QUATERNION and this is the only way the global_position gets set.  The HITL simulation doesn't work without that.  Would this be the right code to add ?

        GNSSReport gnss = sensors.getGNSS();
        if (gnss != null) {
            MAVLinkMessage msg_quaternion = new MAVLinkMessage(schema, "HIL_STATE_QUATERNION", sysId, componentId);
            msg_quaternion.set("time_usec", tu);
            Quat4d quaternion = new Quat4d();
            quaternion.set(vehicle.getRotation());
            Object[] quat = new Object[4];
            quat[0] = (float) quaternion.getW();
            quat[1] = (float) quaternion.getX();
            quat[2] = (float) quaternion.getY();
            quat[3] = (float) quaternion.getZ();
            msg_quaternion.set("attitude_quaternion", quat);
            Vector3d rotationRate = vehicle.getRotationRate();
            msg_quaternion.set("rollspeed", (float) rotationRate.getX());
            msg_quaternion.set("pitchspeed", (float) rotationRate.getY());
            msg_quaternion.set("yawspeed", (float) rotationRate.getZ());
            msg_quaternion.set("lat", (long) (gnss.position.lat * 1e7));
            msg_quaternion.set("lon", (long) (gnss.position.lon * 1e7));
            msg_quaternion.set("alt", (long) (gnss.position.alt * 1e3));
            msg_quaternion.set("vx", (int) (gnss.velocity.x * 100));
            msg_quaternion.set("vy", (int) (gnss.velocity.y * 100));
            msg_quaternion.set("vz", (int) (gnss.velocity.z * 100));
            tv = sensors.getAcc();
            msg_quaternion.set("xacc", tv.x);
            msg_quaternion.set("yacc", tv.y);
            msg_quaternion.set("zacc", tv.z);
            sendMessage(msg_quaternion);
        }

I've tried it, and the drone now takes off when I type "commander takeoff", but it flips over...

lovettchris, 2016-10-18:
This is no longer needed, in fact it creates problems now.

julianoes, 2018-09-13:
I see `HIL_STATE_QUATERNION` in [src/me/drton/jmavsim/Simulator.java](src/me/drton/jmavsim/Simulator.java) and [src/me/drton/jmavsim/MAVLinkHILSystem.java](src/me/drton/jmavsim/MAVLinkHILSystem.java). I'm assuming this is resolved and closing this issue.

closed 2018-09-13 by julianoes