PX4/jMAVSim #35: Run JmavSim in HITL Mode

anujkumargupta, 2016-09-22:
I am trying to run Jmavsim in HITL mode. I follwed the link
https://pixhawk.org/dev/hil/jmavsim
and used /dev/ttyACM0(Pixhawk connected through usb(serial)) and /dev/ttyUSB0(Pixhawk connected through radio telemetry(udp)) and as mention in link ttyusbmodem1 but every time it throws serial port exception either port not found or port busy.

Thanks.

dagar, 2016-09-22:
Is QGroundControl running?

anujkumargupta, 2016-09-22:
Yes its running. I followed the same instructions as mentioned in link in my last comment.

dagar, 2016-09-22:
It's probably QGroundControl autoconnecting first. Kill QGroundControl, start jmavsim, then start QGC (if needed).

anujkumargupta, 2016-09-23:
Thanks , Now it does not throw that error but its not showing any mavlink messages in terminal. Even its not showing the init mavlink.

dagar, 2016-09-23:
What do you get?

[image]

anujkumargupta, 2016-09-23:
I got this.
[image]

dagar, 2016-09-23:
I don't think I quite follow how you've connected with /dev/ttyUSB0 and the telemetry radio.

Can you try with only the usb connection? /dev/ttyACM0

anujkumargupta, 2016-10-19:
Hi , I am trying to do hitl simulation using the snapdragon flight using the command 
java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -serial /dev/ttyUSB0 921600 -qgc 
but facing problem , jmavsim is starging but i didnot get any mavlink msg
Do i need to change anything in my command( i am confused about 921600)
Thanks

salim616, 2017-02-17:
@DrTon  I have tuned my quad on JmavSim and works very weel on Simulation. the problem is when i write the PID parameters on Pixhwak to test flight . the quad behaves weirdly please find the link for the video : https://www.dropbox.com/s/chxxrwyg2n9kuug/video-1487278360.mp4?dl=0

julianoes, 2018-09-13:
This seems outdated, closing.

closed 2018-09-13 by julianoes