PX4/jMAVSim #39: jMAVSim is crashing

mahesha-path, 2016-10-18:
Hi,

Today i setup qGroundControl and jMAVSim for pixhawk(px4) in Ubuntu 14.04.

When i add target points and set mission mode.It is not working fine in jMAVSim and autopilot crash after some time.

**Below is consol log.**
WARN  [commander] MANUAL CONTROL LOST (at t=113362ms)
WARN  [commander] MANUAL CONTROL LOST (at t=115504ms)
WARN  [commander] MANUAL CONTROL LOST (at t=117348ms)
WARN  [navigator] offboard mission updated: dataman_id=1, count=3, current_seq=0
INFO  [tone_alarm] neutral
WARN  [commander] MANUAL CONTROL LOST (at t=119345ms)
WARN  [commander] MANUAL CONTROL LOST (at t=121478ms)
INFO  [tone_alarm] positive
WARN  [commander] MANUAL CONTROL LOST (at t=123749ms)
WARN  [commander] MANUAL CONTROL LOST (at t=125892ms)
WARN  [commander] MANUAL CONTROL LOST (at t=129342ms)
WARN  [commander] MANUAL CONTROL LOST (at t=130894ms)
WARN  [commander] MANUAL CONTROL LOST (at t=132410ms)
WARN  [commander] MANUAL CONTROL LOST (at t=134089ms)
WARN  [commander] MANUAL CONTROL LOST (at t=135839ms)
INFO  [commander] home: 47.3977420, 8.5455939, 488.00
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977420, 8.5455939, 488.00
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected
WARN  [commander] MANUAL CONTROL LOST (at t=137597ms)
WARN  [commander] MANUAL CONTROL LOST (at t=139410ms)
WARN  [commander] MANUAL CONTROL LOST (at t=141414ms)
WARN  [commander] MANUAL CONTROL LOST (at t=143238ms)
WARN  [commander] MANUAL CONTROL LOST (at t=144999ms)
WARN  [commander] MANUAL CONTROL LOST (at t=146845ms)
WARN  [commander] MANUAL CONTROL LOST (at t=148650ms)
WARN  [commander] MANUAL CONTROL LOST (at t=150552ms)
WARN  [commander] MANUAL CONTROL LOST (at t=152247ms)
WARN  [commander] MANUAL CONTROL LOST (at t=154710ms)
WARN  [commander] MANUAL CONTROL LOST (at t=156825ms)
WARN  [commander] MANUAL CONTROL LOST (at t=158822ms)
WARN  [commander] MANUAL CONTROL LOST (at t=160606ms)
WARN  [commander] MANUAL CONTROL LOST (at t=162400ms)
WARN  [commander] MANUAL CONTROL LOST (at t=164447ms)
WARN  [commander] MANUAL CONTROL LOST (at t=166419ms)
WARN  [commander] MANUAL CONTROL LOST (at t=168315ms)
WARN  [commander] MANUAL CONTROL LOST (at t=170106ms)
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 33899 us
WARN  [commander] MANUAL CONTROL LOST (at t=172098ms)
WARN  [commander] MANUAL CONTROL LOST (at t=174060ms)

Regards,
Prashant

bkueng, 2016-10-26:
Do you run HITL or SITL?

julianoes, 2018-09-13:
Outdated, closing.

closed 2018-09-13 by julianoes