PX4/jMAVSim #42: Low mass is causing UAV to climb on ARM

sytelus, 2016-10-20:
Recently @kd0aij made a change to set the default mass to 0.8 from 1.0. See below line:

https://github.com/PX4/jMAVSim/blob/master/src/me/drton/jmavsim/Simulator.java#L309

To reproduce issue:

1. Connect Pixhawk board to PC via USB.
2. Start jMavSim in HIL mode using the command line: `java -Djava.ext.dirs= -cp lib/*;out/production/jmavsim.jar me.drton.jmavsim.Simulator -serial COM4 115200 -qgc'
3. Start qGroundControl and arm.
4. Watch quadcopter climb just on arming.
5. Change the mass to 1 or 1.2. Repeat the steps above to see vehicle doesn't climb on arm.

Log file:
[low_mass.zip](https://github.com/PX4/jMAVSim/files/540710/low_mass.zip)

kd0aij, 2016-10-20:
Can you provide a log generated by sdlog2 or the new logger?

bkueng, 2016-10-20:
I was able to reproduce, it climbes with HITL, but not with SITL.
But I'm not sure if changing the mass is the right solution. @kd0aij can you reproduce?

kd0aij, 2016-10-20:
Do you get no flight log with HIL? It would be less work for me to analyze a log than to set up a HIL run.

bkueng, 2016-10-20:
I thought you had this working. Here's a log: http://logs.uaventure.com/view/ZXXowDSpCugMJXjoLfHN4X

kd0aij, 2016-10-20:
Wow, that log has no data on thrust, RC inputs or actuator outputs. As far as I can tell nothing should be happening at all.

kd0aij, 2016-10-20:
@sytelus I don't know how you're commanding the thrust level in your HIL setup, but the mavlink log shows that the only non-zero actuator_control_target is 0.68. If that's the thrust level, then it's no wonder that the quad is taking off.

sytelus, 2016-10-20:
I'm not doing anything about commanding the thrust level. My setup is everything with its defaults. After launching jMavLink, I just go to qGroundControl and arm the vehicle. That's all I do. The problem is reproducible even if you don't connect RC to pixhawk board.

One thing that is different in HIL and SITL is airframe. For HIL, I set the Airframe to Simulation > HIL Quadcopter X. For SITL, I think the default airframe is something else.

kd0aij, 2016-10-20:
Then neither one of us knows what thrust level has been commanded. Why do you expect it to stay on the ground then?

sytelus, 2016-10-20:
As I'd mentioned, I'm using qGroundControl to arm the vehicle by sliding the switch. There is nothing else I'm doing to send any thrust commands. My setup is simple HIL with everything default. The only additional thing is setting airframe to Simulation > HIL Quadrotor X as per the readme instructions. It is clear that problem is not occurring without your change and is occurring with your change. I would suggest to test your changes with HIL first and then push your changes.

mhkabir, 2016-10-22:
The the difference here is that @sytelus and the others don't have an RC unit or other

julianoes, 2016-10-22:
This is nothing about a wrong mass but the issue is that the actuator scaling is wrong (or at least different for SITL vs. HITL right now.

A fix is coming up: https://github.com/PX4/jMAVSim/pull/45, https://github.com/PX4/Firmware/pull/5716.

LorenzMeier, 2016-10-28:
Fixed.

closed 2016-10-28 by LorenzMeier