PX4/jMAVSim #43: Recent implementation for HIL_STATE_QUATERNION is causing quite a few issues

sytelus, 2016-10-20:
Recently user @kd0aij added HIL_STATE_QUATERNION message along with few other changes such as removing nav_mode check. This is causing following issues:

1. With all default HIL setup (see [steps in this issue](https://github.com/PX4/jMAVSim/issues/42) if needed for repro) you will flickers in qGroundControl where Altitude keeps switching between 0 and 480m.
2. With RC control, you will frequently see quadcopter suddenly flip over and dive to ground.
3. You may see heading for the vehicle at 90 degrees instead of 0.

The cause of this is possibly below issues in the implementation:
1. altitude (alt) is set to z value of vehicle position. This is incorrect. The z position needs to be projected on global frame to get altitude.
2. The lat and lon fields of message are not set at all.
3. The ind_windspeed is not set at all (this might be ok, however).
4. The quaternion has some issue which is causing wrong heading.

Additionally, I also think this message is _possibly_ interfering  PX4 autopilot. See the function `MavlinkReceiver::handle_message_hil_state_quaternion(mavlink_message_t *msg)` in the PX4 code.

I reverted the MAVLinkHILSystem.java file to version [before Sep 1](https://github.com/PX4/jMAVSim/commit/744585dfa14fb0d51fda665fd32cb66a182ef868#diff-8b8d489c12560ab9c476a1f4c02239ce) and above problems went away with vehicle having stable flight.

sytelus, 2016-10-20:
Looks like @lovettchris already tried this change https://github.com/PX4/jMAVSim/pull/41

sytelus, 2016-10-26:
@kd0aij, @bkueng I think I now understand the purpose of HIL_STATE_QUATERNION . So basically, HIL can be used in two modes: (1) "state-level" mode (2) "sensor-level" mode. In Mode 1, pose estimation is disabled. This mode is useful for high throughput simulation where you don't care about errors from sensors. Mode 2 is normal simulation mode where pose estimation is enabled. Mode 2 should be the default and should be used majority of the times. So basically jMavSim *should* provide a command line switch to select one of these two modes instead of always sending HIL_STATE_QUATERNION all the time as it is doing now and thus always enforcing Mode 1.

Reference: https://pixhawk.org/dev/hil/start

bkueng, 2016-10-27:
The reference is outdated and I don't think we support state-level HIL anymore. As I wrote in https://github.com/PX4/jMAVSim/pull/41#issuecomment-255348808, HIL_STATE_QUATERNION is only used for logging.

julianoes, 2018-09-13:
@bkueng do we still need this for logging? #82 would not be needed then.

bkueng, 2018-09-13:
Yes it is, as it provides the ground-truth.

julianoes, 2018-09-13:
Ah ok.

closed 2018-09-13 by julianoes