PX4/jMAVSim #59: jMAVSim without GUI

GregoireH, 2017-05-05:
Hello,

Thanks to this post: http://discuss.px4.io/t/lightweight-or-command-line-sitl-simulation-options/2285
Modifying Visualizer3D.java3, and change setVisible(true) to 
setVisible(false).

I could run SITL using jMAVSim without the GUI as my low computational power computer was having troubles running it properly.

Do you know if most of PX4 users are using the graphics ?
Or if most of us are just wasting CPU with jMAVSim fency graphics...

I was wondering if we could set a parameter for this in order to be able to specify that we want to turn off graphics, without having to edit and compile jMAVSim .
Do you think it could be feasible and that some people could benefit from that ?

looking forward hearing from you.
Best

julianoes, 2017-05-05:
Hi, thanks for the note. I think this would be worthwhile to add. Possibly it should be consistent with the `HEADLESS=1` mode that you can use for the Gazebo simulation.

julianoes, 2018-09-13:
This would still be nice, someday. FYI @JonasVautherin.

JonasVautherin, 2018-09-13:
If I remember correctly, I tried that a few months ago. But the fix proposed here was only disabling the graphics rendering and not making it truly headless (somehow it still needed an X server to be running, so I could not run that, say, in a headless docker).

ekatzfey, 2019-08-10:
Did this ever get resolved? I would really like to be able to run this in a headless docker.

Thanks!

hamishwillee, 2019-08-12:
@ekatzfey No, it did not. We'd happily accept a PR! 

Note, FWIW
-  you can run _Gazebo_ headless: http://dev.px4.io/master/en/simulation/gazebo.html#headless-mode
- With [lockstep](http://dev.px4.io/master/en/simulation/#lockstep-simulation) now supported the simulation may still run fairly "heavy" but it is much more reliable on low performance hardware.

katzfey, 2019-08-18:
I submitted a PR.

PR #105 (2019-08-22): jMAVSim: Added the option to run without the GUI
Added a new command line argument to allow the user to run without the GUI. The
new argument is -no-gui. This allows use of the simulator in environments that
do not have a GUI, like a headless Docker.

Fixes #59.
closed 2019-08-22 by julianoes