PX4/jMAVSim #69: It doesn't work on RX560

Vavooon, 2018-02-14:
Here is the output:
```
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x2ddc6a04, isOwner false, <542867c8, 75028136>[count 0, qsz 0, owner <NULL>]]]
        at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
        at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
        at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
        ... 2 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped 
        at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2047)
        at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1917)
        at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1866)
        at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1835)
        at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:79)
        at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:229)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:215)
        at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2272)
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:1020)
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:1049)
        at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:802)
        at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:132)
        at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
        at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:837)
        at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:274)
        at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:154)
        at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
        at me.drton.jmavsim.Simulator.main(Simulator.java:678)
        ... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7fb424001550, refCount 1, unCloseable false]

```

julianoes, 2018-09-10:
What is RX560?
I'm assuming the problem here is that there is no screen attached. That's currently required and a known limitation.

julianoes, 2018-09-12:
Hm. And that's with latest master? And that's not with vmware or VirtualBox? So not the same as https://github.com/DrTon/jMAVSim/issues/42 ?

Vavooon, 2018-09-12:
Now latest master works, thanks.

julianoes, 2018-09-13:
Cool, thanks for reporting back.

closed 2018-09-13 by julianoes