PX4/jMAVSim #72: Pre flight checks required

vrsh, 2018-03-01:
Hey! Currently I am running jMAVSim with the PX4 serially following instructions mentioned in [jMAVSim/HIL](https://pixhawk.org/dev/hil/jmavsim).

My team and I are interested in verifying a quadcopter mathematical model written in Modelica by connecting it to PX4 via serial MAVLink and viewing the same in QGCS. We could send data using HIL_SENSOR, HIL_GPS and receive controller output using HIL_ACTUATOR_CONTROLS, but the controller output we are receiving is an array of zeros. We think pre-flight checks are not being satisfied for the PX4 in order for it to return proper control output data. I looked into the jMAVSim code to identify these checks, unfortunately I couldn't find something concrete w.r.t the checks. 

Please let me know what these pre-flight checks are or where I could find them or the variable values that need to set in order for the receive valid actuator values.

bkueng, 2018-03-01:
The preflight checks are on the vehicle: https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp.
If you connect QGC at the same time and try to arm, you should see why it does not work.

vrsh, 2018-03-01:
@bkueng  Thanks, but the checks are in .cpp , how to enable/disable it via MAVLink commands?

bkueng, 2018-03-01:
Well, the checks are there for a reason and cannot simply be circumvented. You need to find out why the checks are failing, otherwise your HIL setup will probably not work as expected.

closed 2018-09-10 by julianoes