PX4/jMAVSim #80: Warnings with Java 10

julianoes, 2018-09-11:
With Java 10, we get a bunch of warnings and errors:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "AWT-EventQueue-0" java.lang.StackOverflowError
	at java.desktop/java.awt.Font.getStringBounds(Font.java:2603)
	at java.desktop/java.awt.Font.getStringBounds(Font.java:2517)
	at java.desktop/javax.swing.text.WrappedPlainView.updateMetrics(WrappedPlainView.java:445)
	at java.desktop/javax.swing.text.WrappedPlainView.setSize(WrappedPlainView.java:511)
	at java.desktop/javax.swing.plaf.basic.BasicTextUI$RootView.setSize(BasicTextUI.java:1815)
	at java.desktop/javax.swing.plaf.basic.BasicTextUI.getPreferredSize(BasicTextUI.java:942)
	at java.desktop/javax.swing.plaf.basic.BasicTextAreaUI.getPreferredSize(BasicTextAreaUI.java:121)
	at java.desktop/javax.swing.JComponent.getPreferredSize(JComponent.java:1680)
	at java.desktop/javax.swing.JTextArea.getPreferredSize(JTextArea.java:609)
	at java.desktop/javax.swing.text.JTextComponent.getScrollableTracksViewportHeight(JTextComponent.java:2107)
	at java.desktop/javax.swing.ScrollPaneLayout.layoutContainer(ScrollPaneLayout.java:812)
	at java.desktop/java.awt.Container.layout(Container.java:1535)
	at java.desktop/java.awt.Container.doLayout(Container.java:1524)
	at java.desktop/java.awt.Container.validateTree(Container.java:1720)
	at java.desktop/java.awt.Container.validate(Container.java:1655)
	at java.desktop/sun.awt.X11.XTextAreaPeer.setFont(XTextAreaPeer.java:393)
	at java.desktop/java.awt.Component.validate(Component.java:2934)
	at java.desktop/java.awt.Container.validateTree(Container.java:1738)
	at java.desktop/java.awt.Container.validate(Container.java:1655)
	at java.desktop/sun.awt.X11.XTextAreaPeer.setFont(XTextAreaPeer.java:393)
```

julianoes, 2018-09-13:
It turns out that on macOS this crashes:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00000001096ac0cc, pid=62391, tid=775
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.4+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.4+11, mixed mode, tiered, compressed oops, g1 gc, bsd-amd64)
# Problematic frame:
# C  [libosxapp.dylib+0x20cc]  -[NSApplicationAWT sendEvent:]+0x179
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/julianoes/src/Firmware/Tools/jMAVSim/out/production/hs_err_pid62391.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
/Users/julianoes/src/Firmware/Tools/jmavsim_run.sh: line 47: 62391 Abort trap: 6           java -XX:GCTimeRatio=20 -Djava.ext.dirs= -jar jmavsim_run.jar $device $extra_args
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x000000010f8730cc, pid=62393, tid=775
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.4+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.4+11, mixed mode, tiered, compressed oops, g1 gc, bsd-amd64)
# Problematic frame:
# C  [libosxapp.dylib+0x20cc]  -[NSApplicationAWT sendEvent:]+0x179
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/julianoes/src/Firmware/Tools/jMAVSim/out/production/hs_err_pid62393.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

julianoes, 2018-09-13:
Looks the same as this:
https://bugs.openjdk.java.net/browse/JDK-8197910

gortium, 2019-05-12:
Same thing with 11.

Does it work with 9?

julianoes, 2019-05-13:
As far as I know it works with Java 8.

julianoes, 2024-03-04:
Don't care anymore, haven't used this in a long time.

closed 2024-03-04 by julianoes