PX4/jMAVSim #81: Crash with macOS and Java 9

julianoes, 2018-09-13:
On macOS with Java 9 we get the follow segfault:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00000001096ac0cc, pid=62391, tid=775
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.4+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.4+11, mixed mode, tiered, compressed oops, g1 gc, bsd-amd64)
# Problematic frame:
# C  [libosxapp.dylib+0x20cc]  -[NSApplicationAWT sendEvent:]+0x179
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/julianoes/src/Firmware/Tools/jMAVSim/out/production/hs_err_pid62391.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
/Users/julianoes/src/Firmware/Tools/jmavsim_run.sh: line 47: 62391 Abort trap: 6           java -XX:GCTimeRatio=20 -Djava.ext.dirs= -jar jmavsim_run.jar $device $extra_args
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x000000010f8730cc, pid=62393, tid=775
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.4+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.4+11, mixed mode, tiered, compressed oops, g1 gc, bsd-amd64)
# Problematic frame:
# C  [libosxapp.dylib+0x20cc]  -[NSApplicationAWT sendEvent:]+0x179
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/julianoes/src/Firmware/Tools/jMAVSim/out/production/hs_err_pid62393.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

```
---------------  S U M M A R Y ------------

Command Line: -XX:GCTimeRatio=20 -Djava.ext.dirs= jmavsim_run.jar -udp 127.0.0.1:14560 -r 500

Host: MacBookPro12,1 x86_64 2700 MHz, 4 cores, 8G, Darwin 17.7.0
Time: Thu Sep 13 11:18:10 2018 CEST elapsed time: 6 seconds (0d 0h 0m 6s)

---------------  T H R E A D  ---------------

[... 122 lines cut ...]
  0x00007f8381838800 ConcurrentGCThread "G1 Marker#0" [stack: 0x000070000a54a000,0x000070000a64a000] [id=13571]
  0x00007f8381008800 ConcurrentGCThread "G1 Refine#0" [stack: 0x000070000a241000,0x000070000a341000] [id=19715]
  0x00007f8381008000 ConcurrentGCThread "G1 Refine#1" [stack: 0x000070000a13e000,0x000070000a23e000] [id=19971]
  0x00007f8380803000 ConcurrentGCThread "G1 Refine#2" [stack: 0x000070000a03b000,0x000070000a13b000] [id=20483]
  0x00007f8381812800 ConcurrentGCThread "G1 Refine#3" [stack: 0x0000700009f38000,0x000070000a038000] [id=12291]
  0x00007f8381009800 ConcurrentGCThread "G1 Young RemSet Sampling" [stack: 0x000070000a344000,0x000070000a444000] [id=19459]

Threads with active compile tasks:
C2 CompilerThread11799       4       java.lang.invoke.InvokerBytecodeGenerator::emitImplicitConversion (159 bytes)
C2 CompilerThread01837       4       java.util.jar.JarFile::getEntry (60 bytes)

VM state:not at safepoint (normal execution)

VM Mutex/Monitor currently owned by a thread: None

Heap address: 0x0000000740000000, size: 2048 MB, Compressed Oops mode: Zero based, Oop shift amount: 3
Narrow klass base: 0x0000000000000000, Narrow klass shift: 3
Compressed class space size: 1073741824 Address: 0x00000007c0000000
```

Looks the same as this:
https://bugs.openjdk.java.net/browse/JDK-8197910

More workarounds and tries that didn't work for me:
http://forum.jogamp.org/Jogl-Jogamp-on-Java-9-tt4038012.html

DaeunGod, 2018-10-15:
Hi julianoes, Did you solve this problem? because i have a same problem too. 

i tried to run jmavsim_run.jar with java7 and java8 but it showed the different errors.

So, if you have any advice then please notice me.

thanks.

julianoes, 2018-10-15:
It works with Java 8. Make sure to clean build by deleting the `out` folder, and otherwise post your error.

DaeunGod, 2018-10-15:
@julianoes Thanks. i will try it!

julianoes, 2018-12-02:
The steps for the Java 8 workaround are:

```
cd wherever/Firmware
rm -rf Tools/jMAVSim/out
export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)
make posix jmavsim
```

Also, see: http://discuss.px4.io/t/problem-running-jmavsim-on-macos-high-sierra/5357/20

nicollorens12, 2018-12-02:
Hi! Thanks for the response,

Now i'm getting this error:
Exception in thread 'main' java.lang.reflect.InvocationTargetException
followed by many other lines of errors...

julianoes, 2018-12-03:
@nicollorens12 could it be that you ran the simulation in a new shell window after doing the steps I mentioned above? The `export` command is only valid for the current shell, and it could be that you're seeing these errors because it is using Java 11 to run the Java 8 compiled code.

nicollorens12, 2018-12-03:
I'm getting this error with java 11 but also with java 8. I haven't tried
any export comand. In the instalation guide doesn't say you have to do any
export comand.

nicollorens12, 2018-12-03:
> @nicollorens12 <https://github.com/nicollorens12> you need to follow the
> steps I posted above.

Oh sorry, I'll try later when I get home. Thank you so much. I'll tell you
the result I get.

nicollorens12, 2018-12-03:
Hi again,

I'm trying the export comand but I get this error. bash:
/usr/libexec/java_home: No such file or directory

julianoes, 2018-12-03:
> I'm trying the export comand but I get this error. bash: /usr/libexec/java_home: No such file or directory

Ok, this probably means that you don't have Java 8 installed. I'd try to install Java 8.

nicollorens12, 2018-12-03:
Hi,

I've modified the export comand to the path lf my java, now i'm getting
this back.

Unrecognized option: -v
Error: Could not create the Java Virtual Machine
Error: A fatal exception occurred. Program will exit.

I do java 8 install because when I java -version I get openjdk version
1.8.0_191 and everything.

julianoes, 2018-12-03:
Have you tried `-version` or `--version`?

nicollorens12, 2018-12-03:
I've tried again the comand make posix jmavsim and now I get something to
work!! It says Waiting for initial data on UDP port 14560. Please start the
flight simulator to proceed!

But shouldn't I get the jmavsim program opened already?

julianoes, 2018-12-03:
Look through the output, something must have failed if you see that.

nicollorens12, 2018-12-03:
Yes I see. After saying px4 started it says:
/home/nico/Firmware/Tools/jmavsim_run.sh: line 58: ant: comand not found

julianoes, 2018-12-03:
https://stackoverflow.com/questions/3222804/how-can-i-install-apache-ant-on-mac-os-x/20984975#20984975
```
brew install ant
```

nicollorens12, 2018-12-03:
I'm getting now the same. The build is succesful but when it starts the GUI
it says in thread "main" java.lang.reflect.InvocationTargetException

I do have java 8

julianoes, 2018-12-04:
@nicollorens12 can you please do `make clean` and then build it and copy all of the output to pastebin and link it here?

nicollorens12, 2018-12-07:
Sorry for the late respond

Pastebin: pastebin.com/W4jW5ptt

julianoes, 2018-12-07:
I would need "all of the output" including `make clean`, the exports, etc.

nicollorens12, 2018-12-07:
Here you have it: pastebin.com/CSVSvLCy

nicollorens12, 2018-12-08:
> Ok, so the export command didn't work, so of course jMAVSim won't work
> properly later.
>
> export JAVA_Home=$(/usr/libexec/java_home -v 1.8)
> bash: /usr/libexec/java_home: No such file or directory
>
> I would try to install Oracle Java SE Development Kit 8:
>
> https://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html
>
> And on the side, why are you doing these things as root? That seems
> dangerous.

I'm reinstalling ubuntu because I think I'm messing up the enviroment
paths. I'll install a clean java 8 version and try to build the program.
Thanks

julianoes, 2018-12-08:
Oh, and I was assuming and trying to help with macOS!

This issue is about macOS not Ubuntu. For Ubuntu I suggest you create a new issue, that way it gets clearer.

julianoes, 2020-03-02:
This one has been resolved, see: https://dev.px4.io/master/en/simulation/jmavsim.html#an-illegal-reflective-access-operation-has-occured

closed 2020-03-02 by julianoes