PX4/jMAVSim #84: visualisation: No trees or skies under Win 10

copterspace, 2018-11-10:
Hello, dear friends.
I run jmavsim on Windows 10 Cygwin toolchain, according to this instrustion:
http://dev.px4.io/en/setup/dev_env_windows_cygwin.html 
It runs OK (takeoff, hold, land, QGS connect, etc), but visualisation window shows only drone and ground - no trees, skies, other evironment - only black background instead:
[image]

When ran on Ubuntu before - I never had such an issue.
Can you please suggest me how to make visualisation work correctly under Windows 10?
It would be better to see at least constant blue sky, instead of black one.

MaEtUgR, 2018-11-13:
**Confirmed**
@copterspace Thanks for the report! I honestly also have that but only on part of the computers I'm using. The problem seems to be in the java graphics library that jMAVsim uses.

**History**
In https://github.com/PX4/Firmware/pull/8194 a higher resolution background was introduced and that seems to break some memory threshold of the java graphics library for certain graphics drivers. It already happened inside VMWare also for Ubuntu and we have a fix with the specific VMWare driver here: https://github.com/PX4/jMAVSim/pull/63

At home on my GTX 1080 the bakcground shows normal for some reason... but I added it to the Cygwin todo list 👍

obbe79, 2020-03-07:
Same problem here. After installing the Cygwin toolchain I get the black sky on Win10. On a different PC with Win7 I see JMAVSim correctly

obbe79, 2020-03-09:
> Same problem here. After installing the Cygwin toolchain I get the black sky on Win10. On a different PC with Win7 I see JMAVSim correctly

To temporarily solve the problem I changed line 276 in Visualizer3D.java to:

`            tex = loadTexture(TEX_DIR + SKY_TEXTURE_LOW_RES);`

to force the low res texture in any case.

copterspace, 2020-03-10:
Yes,  SKY_TEXTURE_LOW_RES helped, now looks quite good:
[image]

I think, it is possible to use SKY_TEXTURE_LOW_RES on all platforms. Or detect Win10 in Java somehow?...