PX4/jMAVSim #85: Can't get jMAVSim to work

nicollorens12, 2018-12-02:
Hi guys,
I've been literally trying the whole weekend to get jMAVSim to work but it's been imposible (probably because of my very basic knowledge). When I try to execute the comand:
`make px4_sitl jmavsim` 
I get px4 to start but then (if i'm not mistaken) it tries to open jMAVSim but this error occurs:
`Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
	at java.awt.Window.getToolkit(Window.java:1358)
	at java.awt.Window.init(Window.java:506)
	at java.awt.Window.<init>(Window.java:537)
	at java.awt.Frame.<init>(Frame.java:420)
	at java.awt.Frame.<init>(Frame.java:385)
	at javax.swing.JFrame.<init>(JFrame.java:189)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:104)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
	... 5 more

`
Does anyone know how to fix this? I've tried so hard but I can't seem to get it working. Thanks in advanced

hamishwillee, 2018-12-02:
It probably isn't you. What setup instructions did you follow and to what platform? Are you running in a VM?

nicollorens12, 2018-12-03:
Hi, I've used this instructions.
https://github.com/PX4/jMAVSim/blob/master/README.md
But I couldn't get the comand ant create_copy_res working so I tried
directly with the comand make px4_sitl jmavsim using the px4 developer
guide.

I'm using ubuntu, but not in a VM.

hamishwillee, 2018-12-03:
The setup instructions here worked for me last time I tried on Ubuntu 16.04: http://dev.px4.io/en/setup/dev_env_linux.html (there are helper scripts there that install everything needed).

The problem is probably due to an incompatible Java installation.

Which is not to say that there isn't a problem - those README instructions should also be updated.

nicollorens12, 2018-12-03:
> OpenJDK8 according to to the docs:

Should I uninstall first the java version that i'm using now?

hamishwillee, 2018-12-03:
Probably. I just know what works as a turnkey installation on a clean ubuntu 16.04 machine.

J4NSS3NS, 2019-11-19:
I saw different issues regarding java or wrong java versions on px4 builds Jmavsim.
I had different problems on Arch system but because system was used for trial and error I made fresh ubuntu 18 install. 
I am unable to make it work, I did not save all of the error messages regarding Java because I just wanted px4 to work.

I noticed strange thing: I installed
nico@nico-MacBookPro:~$ sudo apt-get install ant openjdk-8-jdk openjdk-8-jre -y
went back i Terminal to check , because for Jmavsim prerequists those were needed....

After I sourced the ubuntu_sim.sh script, the Java version was

nico@nico-MacBookPro:~/src/Firmware$ java -version
openjdk version "11.0.4" 2019-07-16
OpenJDK Runtime Environment (build 11.0.4+11-post-Ubuntu-1ubuntu218.04.3)
OpenJDK 64-Bit Server VM (build 11.0.4+11-post-Ubuntu-1ubuntu218.04.3, mixed mode, sharing)

Is it possible problems arise in reinstallating java through script file?

error now on my machine is :

Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!