PX4/jMAVSim #90: "Can't load library: /home/pi/src/Firmware/Tools/jMAVSim/out/production/libgluegen-rt.so"

efischer91, 2019-01-22:
I followed these instructions: 
https://github.com/PX4/jMAVSim/blob/master/README.md

On running it with 
`java -djava.ext.dirs= -jar jmavsim_run.jar`

I get the following error:
````
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.UnsatisfiedLinkError: Can't load library: /home/pi/src/Firmware/jMAVSim/out/production/libgluegen-rt.so
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1827)
	at java.lang.Runtime.load0(Runtime.java:809)
	at java.lang.System.load(System.java:1086)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibraryInternal(JNILibLoaderBase.java:596)
	at com.jogamp.common.jvm.JNILibLoaderBase.access$000(JNILibLoaderBase.java:63)
	at com.jogamp.common.jvm.JNILibLoaderBase$DefaultAction.loadLibrary(JNILibLoaderBase.java:95)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibrary(JNILibLoaderBase.java:459)
	at com.jogamp.common.os.DynamicLibraryBundle$GlueJNILibLoader.loadLibrary(DynamicLibraryBundle.java:421)
	at com.jogamp.common.os.Platform$1.run(Platform.java:317)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.common.os.Platform.<clinit>(Platform.java:287)
	at com.jogamp.opengl.GLProfile.<clinit>(GLProfile.java:146)
	at jogamp.opengl.ThreadingImpl$1.run(ThreadingImpl.java:83)
	at jogamp.opengl.ThreadingImpl$1.run(ThreadingImpl.java:66)
	at java.security.AccessController.doPrivileged(Native Method)
	at jogamp.opengl.ThreadingImpl.<clinit>(ThreadingImpl.java:66)
	at com.jogamp.opengl.Threading.disableSingleThreading(Threading.java:164)
	at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:130)
	at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
	at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:837)
	at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:274)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:155)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:170)
	at me.drton.jmavsim.Simulator.main(Simulator.java:793)
	... 5 more

charles-blouin, 2019-03-01:
I have the same error on a Raspberry Pi with Raspbian stretch.

srmeier, 2019-07-13:
I also cannot load this lib, `Firmware/jMAVSim/out/production/libgluegen-rt.so`, when running `make px4_sitl jmavsim` on the Pinebook.

[image]

nacorti, 2019-10-03:
Also having this same error when running make px4_sitl jmavsim, anyone managed to resolve?

laxmanbobby, 2021-01-19:
@efischer91 have you managed to resolve this? I am experiencing the same issue on Ubuntu 18.04 on a Jetson Nano

efischer91, 2021-01-19:
> @efischer91 have you managed to resolve this? I am experiencing the same issue on Ubuntu 18.04 on a Jetson Nano

Unfortunately no.

alan0054, 2021-06-03:
I have the same error...
I operate at Raspberry Pi 3B board and use Ubuntu mate 18.04
Passed 2 years.. No one can even help us?....

alan0054, 2021-06-11:
> I have the same error...
> I operate at Raspberry Pi 3B board and use Ubuntu mate 18.04
> Passed 2 years.. No one can even help us?....

hey guys ! I find the solution !!!
https://jogamp.org/deployment/archive/rc/v2.4.0-rc-20200104/apk/
Download the gluegen file that matches your device.
My workstation is RPi 3B, so I download the aarch64 version.
After unzip copy "libgluegen_rt.so" file from "gluegen-2.4-b938-20210111-linux-aarch64\gluegen-2.4-b938-20210111-linux-aarch64\lib" to your missing file's folder.(In this path, if there is no matching folder, try to use the command "mkdir + folder name" to generate a new folder to continue copying)

netuserjun, 2021-08-25:
> hey guys ! I find the solution !!!
> https://jogamp.org/deployment/archive/rc/v2.4.0-rc-20200104/apk/
> Download the gluegen file that matches your device.
> My workstation is RPi 3B, so I download the [...]

I did exactly as your way, but I got another problem.
[image]

The error is changed from 'Can't load library' to 'invalid ELF header'.
If you solved this problem, please let me know..
Thanks.

cryptik, 2023-02-21:
I am building jMAVSim for linux/aarch64.  I was able to get past most of the libgluegen_rt issues by downloading the aarch64 jar files from the following location:

[https://jogamp.org/deployment/archive/rc/v2.4.0/jar/](url)

However, while it gets past the initial error indicating that it can't find libraries, with a different one as follows:
```
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.UnsatisfiedLinkError: Couldn't load library 'gluegen_rt' generically including [], nor as [/usr/java/packages/lib/libgluegen_rt.so, /usr/lib64/libgluegen_rt.so, /lib64/libgluegen_rt.so, /usr/lib/libgluegen_rt.so, /usr/lib/libgluegen_rt.so, /home/umex/Documents/tools/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/out/production/libgluegen_rt.so, /home/umex/Documents/tools/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/out/production/natives/linux-aarch64/libgluegen_rt.so]
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibraryInternal(JNILibLoaderBase.java:634)
	at com.jogamp.common.jvm.JNILibLoaderBase.access$000(JNILibLoaderBase.java:64)
	at com.jogamp.common.jvm.JNILibLoaderBase$DefaultAction.loadLibrary(JNILibLoaderBase.java:107)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibrary(JNILibLoaderBase.java:488)
	at com.jogamp.common.os.DynamicLibraryBundle$GlueJNILibLoader.loadLibrary(DynamicLibraryBundle.java:427)
	at com.jogamp.common.os.Platform$1.run(Platform.java:321)
	at com.jogamp.common.util.SecurityUtil.doPrivileged(SecurityUtil.java:80)
	at com.jogamp.common.os.Platform.<clinit>(Platform.java:290)
	at com.jogamp.opengl.GLProfile.<clinit>(GLProfile.java:154)
	at jogamp.opengl.ThreadingImpl$1.run(ThreadingImpl.java:83)
	at jogamp.opengl.ThreadingImpl$1.run(ThreadingImpl.java:66)
	at com.jogamp.common.util.SecurityUtil.doPrivileged(SecurityUtil.java:80)
	at jogamp.opengl.ThreadingImpl.<clinit>(ThreadingImpl.java:66)
	at com.jogamp.opengl.Threading.disableSingleThreading(Threading.java:164)
	at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:129)
	at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
	at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:858)
	at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:267)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:194)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
	at me.drton.jmavsim.Simulator.main(Simulator.java:944)
```

The .so files are there... and I can see them copied during the build, however, I still get the error.  I am hoping this is just a classpath issue, but I am not sure what the next step would be to fix this.

cryptik, 2023-02-21:
Update on the above, the aarch64 jar files are there... but I don't see the .so files in the `./out/production` or `./out/production/natives` folder.  There is no natives folder.  I am not sure how the make file generates this folder and why it's not doing it for the `aarch64` so files.

cryptik, 2023-02-21:
Another update... after copying the right jar files from the `jogamp` website, the build.xml file needs to be updated (...still learning here).  Specifically here:

```
<attribute name="Rsrc-Class-Path" value="./ annotations.jar gluegen-rt-natives-linux-amd64.jar 
gluegen-rt-natives-linux-aarch64.jar gluegen-rt-natives-linux-i586.jar gluegen-rt-natives-macosx-universal.jar
gluegen-rt-natives-windows-amd64.jar gluegen-rt-natives-windows-i586.jar gluegen-rt.jar j3dcore.jar j3dutils.jar
joal-natives-linux-amd64.jar joal-natives-linux-aarch64.jar joal-natives-linux-i586.jar 
joal-natives-macosx-universal.jar joal-natives-windows-amd64.jar joal-natives-windows-i586.jar joal.jar 
jogl-all-natives-linux-amd64.jar jogl-all-natives-linux-aarch64.jar jogl-all-natives-linux-i586.jar 
jogl-all-natives-macosx-universal.jar jogl-all-natives-windows-amd64.jar jogl-all-natives-windows-i586.jar 
jogl-all.jar jssc.jar vecmath.jar"/>
```
and here:

```
<zipfileset dir="lib" includes="gluegen-rt-natives-linux-aarch64.jar"/>
<zipfileset dir="lib" includes="joal-natives-linux-aarch64.jar"/>
<zipfileset dir="lib" includes="jogl-all-natives-linux-aarch64.jar"/>
```
after the above updates, run a `make clean` and then `make px4_sitl_default jmavsim` and the simulator GUI will work just fine.

ai-zeus-hub, 2023-03-24:
Hey @cryptik, I followed your steps above (I'm also using px4) by doing:

* Downloading jars:
```
cd PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/lib/
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/gluegen-rt-natives-linux-aarch64.jar
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/joal-natives-linux-aarch64.jar
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/jogl-all-natives-linux-aarch64.jar
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/joal-natives-linux-aarch64.jar 
```
* Modifying the build.xml file (here is the snippet you referenced):

```
    <target name="create_run_jar" depends="compile" description="Creates standalone runnable JAR file including dependencies.">
        <jar destfile="${build.dir}/jmavsim_run.jar">
            <manifest>
                <attribute name="Main-Class" value="org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader"/>
                <attribute name="Rsrc-Main-Class" value="me.drton.jmavsim.Simulator"/>
                <attribute name="Class-Path" value="."/>
                <attribute name="Rsrc-Class-Path" value="./ annotations.jar gluegen-rt-natives-linux-amd64.jar gluegen-rt-natives-linux-aarch64.jar gluegen-rt-natives-linux-i586.jar gluegen-rt-natives-macosx-universal.jar gluegen-rt-natives-windows-amd64.jar gluegen-rt-natives-windows-i586.jar gluegen-rt.jar j3dcore.jar j3dutils.jar joal-natives-linux-amd64.jar joal-natives-linux-aarch64.jar joal-natives-linux-i586.jar joal-natives-macosx-universal.jar joal-natives-windows-amd64.jar joal-natives-windows-i586.jar joal.jar jogl-all-natives-linux-amd64.jar jogl-all-natives-linux-aarch64.jar jogl-all-natives-linux-i586.jar jogl-all-natives-macosx-universal.jar jogl-all-natives-windows-amd64.jar jogl-all-natives-windows-i586.jar jogl-all.jar jssc.jar vecmath.jar"/>
            </manifest>
            <fileset dir="${classes.dir}"/>
            <zipfileset src="jar-in-jar-loader.zip"/>
            <zipfileset dir="lib" includes="annotations.jar"/>
            <zipfileset dir="lib" includes="gluegen-rt.jar"/>
            <zipfileset dir="lib" includes="j3dcore.jar"/>
            <zipfileset dir="lib" includes="j3dutils.jar"/>
            <zipfileset dir="lib" includes="joal.jar"/>
            <zipfileset dir="lib" includes="jogl-all.jar"/>
            <zipfileset dir="lib" includes="jssc.jar"/>
            <zipfileset dir="lib" includes="vecmath.jar"/>
            <zipgroupfileset dir="lib" includes="forms_rt.jar"/>

            <!-- OS-specific jars (can exclude any you don't need) -->
            <zipfileset dir="lib" includes="gluegen-rt-natives-linux-aarch64.jar"/>
            <zipfileset dir="lib" includes="joal-natives-linux-aarch64.jar"/>
            <zipfileset dir="lib" includes="jogl-all-natives-linux-aarch64.jar"/>

```

* Followed by a clean build:
```
cd PX4-Autopilot/
make clean
make px4_sitl_default jmavsim
```

However, I then run in to a separate error:

```
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 39: [: Illegal number: 
INFO  [init] jMAVSim simulator
INFO  [simulator_mavlink] using TCP on remote host localhost port 4560
WARN  [simulator_mavlink] Please ensure port 4560 is not blocked by a firewall.
INFO  [simulator_mavlink] Resolved host 'localhost' to address: 127.0.0.1
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /home/austinriffle/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:567)
        at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.InternalError: XXX0 profile[2]: GL2 -> profileImpl GL3bc !!! not mapped 
        at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2098)
        at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1976)
        at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1895)
        at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1861)
        at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
        at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:239)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:312)
        at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:225)
        at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2324)
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:1042)
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:1071)
        at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:824)
        at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:131)
        at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
        at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:858)
        at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:267)
        at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:194)
        at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
        at me.drton.jmavsim.Simulator.main(Simulator.java:944)
        ... 5 more
```

Did you (or anyone else for that matter) encounter this `java.lang.InternalError` error, and have any advice for a solution?

ai-zeus-hub, 2023-03-26:
As info, if anyone else runs into the above error, I resolved it by doing:

```
export LIBGL_ALWAYS_SOFTWARE=1
```

according to here:

https://groups.google.com/g/jaamsim-users/c/tJW48U0Vi9k?pli=1

There's probably a way to fix it with a driver update or something similar that will allow us to leverage hardware, but for now this works. If I find another solution I will update.

1qingfeng-1, 2023-05-09:
> 实际上，更新以下jar可以正确解决问题：
> 
> ```
> wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/jogl-all.jar
> wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/gluegen-rt.jar
> wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/joal.jar
> ```

Download the latest build versions of `gluegen`, `joal`, and `jogl` and replace them. Here are the links: https://jogamp.org/chuck/view/fwd/job/gluegen/952/ https://jogamp.org/chuck/view/fwd/job/joal/670/ https://jogamp.org/chuck/view/fwd/job/jogl/1519/

1qingfeng-1, 2023-05-09:
> 实际上，更新以下jar可以正确解决问题：
> 
> ```
> wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/jogl-all.jar
> wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/gluegen-rt.jar
> wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/joal.jar
> ```

Download the latest build versions of ·gluegen, joal, and jogl and replace them. 
Here are the links: 
- https://jogamp.org/chuck/view/fwd/job/gluegen/952/ 
- https://jogamp.org/chuck/view/fwd/job/joal/670/ 
- https://jogamp.org/chuck/view/fwd/job/jogl/1519/

Unzip these three compressed files, 
- replace the relevant JAR files in the directory "PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/lib/"
- add the relevant libraries to the directory "PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/out/production/". 

Now you should be able to run it without any issues.

CapricornFBQ, 2023-07-30:
Thanks above @ ALL  In addition to the steps of @1qingfeng-1. Since I am a CPU of the Mac M series, there are the following changes Used in 
/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/build.xml: 
<zipfileset dir="lib" includes="gluegen-rt-natives-linux-aarch64.jar"/> 
<zipfileset dir="lib" includes="joal-natives-linux-aarch64.jar"/> 
<zipfileset dir="lib" includes="jogl-all-natives-linux-aarch64.jar"/> 
The corresponding resources are as follows: 
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/jogl-all-natives-linux-aarch64.jar 
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/gluegen-rt-natives-linux-aarch64.jar 
wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/joal-natives-linux-aarch64.jar.jar

JasonStepan, 2023-09-05:
> > Actually, updating the following jar's resolve the issue correctly:
> > ```
> > wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/jogl-all.jar
> > wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/gluegen-rt.jar
> > wget https://jogamp.org/deployment/archive/rc/v2.4.0/jar/joal.jar
> > ```
> 
> Download the latest build versions of ·gluegen, joal, and jogl and replace them. Here are the links:
> 
> * https://jogamp.org/chuck/view/fwd/job/gluegen/952/
> * https://jogamp.org/chuck/view/fwd/job/joal/670/
> * https://jogamp.org/chuck/view/fwd/job/jogl/1519/
> 
> Unzip these three compressed files,
> 
> * replace the relevant JAR files in the directory "PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/lib/"
> * add the relevant libraries to the directory "PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/out/production/".
> 
> Now you should be able to run it without any issues.

It did work for me.