PX4/jMAVSim #93: Yuneec 520: HITL simulator keep crashing into land after takeoff

Anhtastic, 2019-02-16:
HITL simulator would work but after taking off it immediately crash into land. It start wobbling as soon as it take off and then crash into land. 

I'm following instructions from here: https://developer.yuneec.com/documentation/121/HITL-Testing-for-Mac

I've made sure the QGC HITL is enabled and closing it before running the simulator.

I'm currently using:
 java version "1.8.0_202"
Java(TM) SE Runtime Environment (build 1.8.0_202-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.202-b08, mixed mode)

Greatly appreciated if someone can help me figure out why the Yuneec 520 drone simulator keeps on crashing after taking off.

hamishwillee, 2019-02-17:
@julianoes FYI - any ideas who might be able to chase this one up?

julianoes, 2019-02-18:
Thanks @hamishwillee.

@potaito @simonegu I can try to reproduce this for now.

potaito, 2019-02-18:
I was unable to use H520 HITL as well after updating to the newest commit. Last known working commit for me is cb2d13087a671d80a68e65109ff4da307022dea0, and the broken commit is 00bdf6d92b6a8678f5b683a3ebca9e52e88cb5ce. I have not yet found the time to look into this :/ There are two dozen commit in between and it should be possible to pin-point when exactly H520 HITL broke.

julianoes, 2019-02-18:
Right, so I must have broken it with the lockstep changes. I'll check.

potaito, 2019-02-18:
Thank you very much

julianoes, 2019-02-18:
I think I've fixed it. Would be cool if you could also test: #94 .

potaito, 2019-02-18:
Tested and confirmed to be working :+1:

closed 2019-02-18 by julianoes

Anhtastic, 2019-02-19:
Sorry guys, it was President's day yesterday so no working day, but I have tested this morning and it worked. Thank you so much! 👍

PR #94 (2019-02-18): Simulator: make lockstep an option
This adds a CLI argument to enable lockstep. This should restore HITL in its original working shape.

The problem was mainly that we used the simulator time instead of the computer time and also we ran update twice which is required for lockstep. One step is to receive messages from the autopilot, and the
other step is to send new messages out.

This change will require that we use the cli flag on the PX4 side.

Fixes #93.