PX4/jMAVSim #96: jMAVsim can't work normally

louzq16, 2019-03-05:
hello everyone!
I tried to run jMAVsim on ubuntu18.04,but it failed to start GUI,
this is the message：
'''
louzq@louzq-HP-Pavilion-Power-Laptop-15-cb0xx:~/src/Firmware/Tools/jMAVSim$ java -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (file:/home/louzq/src/Firmware/Tools/jMAVSim/lib/j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
'''
how to solve this problem

Ahmad-Drak, 2019-04-18:
Same issue. Did you manage to fix it?

Stifael, 2019-04-18:
same issue

Stifael, 2019-04-18:
I assume you upgraded java version.
In case you have `openjdk 11` (`java --version`), I would downgrade to 10 or 9. There was already a fix for 9/10, but not for 11 (https://github.com/PX4/jMAVSim/commit/cb2d13087a671d80a68e65109ff4da307022dea0).

To downgrade your version:
* download jdk-10 from [here](https://jdk.java.net/archive/)
* unzip
```
tar xzvf openjdk-10_linux-x64_bin.tar.gz
```
* move jdk-10 to install directory and update version to 10.
```
sudo mkdir -p /usr/lib/jvm/java-10-openjdk-amd64/
sudo mv jdk-10 /usr/lib/jvm/java-10-openjdk-amd64/
```
sudo update-alternatives --install /usr/bin/java java /usr/lib/jvm/java-10-openjdk-amd64/jdk-10/bin/java 1
sudo update-alternatives --install /usr/bin/javac javac /usr/lib/jvm/java-10-openjdk-amd64/jdk-10/bin/javac 1
sudo update-alternatives --config java
```
* remove jmavsim build folder
```
rm -rf <path-to-px4>/Tools/jMAVSim/out
```

Now you should have your GUI back. 

**NOTE**: this is just a workaround and not a fix at all.

Ahmad-Drak, 2019-04-19:
Thanks for the workaround, will git it a try

Ahmad-Drak, 2019-04-23:
@Stifael jdk-10 didnt work for me, however jdk-9 did work. Thanks for the help

louzq16, 2019-04-23:
> I assume you upgraded java version.
> In case you have `openjdk 11` (`java --version`), I would downgrade to 10 or 9. There was already a fix for 9/10, but not for 11 ([cb2d130](https://github.com/PX4/jMAVSim/commit/cb2d13087a671d80a68e65109ff4da307022dea0)).
> 
> To downgrade your version:
> 
> * download jdk-10 from [here](https://jdk.java.net/archive/)
> * unzip
> 
> ```
> tar xzvf openjdk-10_linux-x64_bin.tar.gz
> ```
> * move jdk-10 to install directory and update version to 10.
> 
> ```
> sudo mkdir -p /usr/lib/jvm/java-10-openjdk-amd64/
> sudo mv jdk-10 /usr/lib/jvm/java-10-openjdk-amd64/
> sudo update-alternatives --install /usr/bin/java java /usr/lib/jvm/java-10-openjdk-amd64/jdk-10/bin/java 1
> sudo update-alternatives --install /usr/bin/javac javac /usr/lib/jvm/java-10-openjdk-amd64/jdk-10/bin/javac 1
> sudo update-alternatives --config java
> ```
> * remove jmavsim build folder
> 
> ```
> rm -rf <path-to-px4>/Tools/jMAVSim/out
> ```
> Now you should have your GUI back.
> 
> **NOTE**: this is just a workaround and not a fix.
thank you for your help!! It works well now~

julianoes, 2019-06-11:
For Ubuntu, you can fall back to Java 8 quite easily:
```
sudo apt install openjdk-8-jdk
sudo update-alternatives --config java # choose 8
rm -rf Tools/jMAVSim/out
```

yuhaim, 2019-07-30:
> For Ubuntu, you can fall back to Java 8 quite easily:
> 
> ```
> sudo apt install openjdk-8-jdk
> sudo update-alternatives --config java # choose 8
> rm -rf Tools/jMAVSim/out
> ```

This solution works for me finally, but there is another issue as follow:

Options parsed, starting Sim.
Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
	at java.awt.Window.getToolkit(Window.java:1358)
	at java.awt.Window.init(Window.java:506)
	at java.awt.Window.<init>(Window.java:537)
	at java.awt.Frame.<init>(Frame.java:420)
	at java.awt.Frame.<init>(Frame.java:385)
	at javax.swing.JFrame.<init>(JFrame.java:189)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:107)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:176)
	at me.drton.jmavsim.Simulator.main(Simulator.java:895)
	... 5 more

This could be solved by [https://askubuntu.com/questions/695560/assistive-technology-not-found-awterror](url)

julianoes, 2019-07-30:
Thanks for adding that, also see: https://github.com/PX4/Firmware/issues/9557#issuecomment-512137607

closed 2019-07-30 by julianoes