PX4/jMAVSim #103: Port busy when run JmavSim in HITL Mode In cygwin

Evaneee, 2019-05-15:
My operation system is windows 
so I use cygwin Tool chain

to run Jmavsim in HITL mode,I followed the link
https://dev.px4.io/en/simulation/hitl.html
but when I type ./Tools/jmavsim_run.sh -q -d /dev/ttyS2 -b 921600 -r 250
every time it throws serial port busy.

need help!

Thanks

[image]

hamishwillee, 2019-05-15:
I don't know if HITL debugging is supported in Cygwin, but @MaEtUgR will.

MaEtUgR, 2019-05-15:
@Evaneee I successfully used HITL with the Cygwin toolchain on Windows before.

1. Make sure the vehicle is in HITL mode with the parameter `SYS_HITL`
1. Make sure QGroundcontrol is closed
1. I think I started HITL with the command `./Tools/jmavsim_run.sh -q -d COM3 -b 921600 -r 250` COM3 is the device I see in your device manager. I'm not sure anymore but I think I had to use the windows naming of the port because it goes through java which is windows native.
1. As soon as it connected successfully you can open QGC again and do your tests

Sorry I can't currently test but can you try and write if it works?

Evaneee, 2019-05-16:
@MaEtUgR 

Very Good,  It works!!!
Thanks a lot!
[image]

hamishwillee, 2019-05-16:
Awesome. I'm closing this. @MaEtUgR I've added note to the docs - can you please check them? https://github.com/PX4/Devguide/pull/791

closed 2019-05-16 by hamishwillee