PX4/jMAVSim #109: px4_task_spawn_cmd: failed to set sched policy

LJ0707, 2020-01-15:
[image]

dagar, 2020-01-15:
This is a px4 platform bug. What's the environment and PX4 version?

bozkurthan, 2020-01-22:
Hello @dagar I'm working on Windows10 and latest build (v1.11.0-beta1-337-g65aaf5170c). I also receive this problem.

bozkurthan, 2020-01-22:
FYI @dagar I checkout with stable version(v1.10.0) and running without problem. 
Here is the outputs:

```
$ git checkout -f tags/v1.10.0
warning: unable to rmdir 'platforms/qurt/dspal': Directory not empty
Checking out files: 100% (2268/2268), done.
Note: checking out 'tags/v1.10.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 4f6faac2c8 ECL fix for v1.10.0 release
```

```
$ git submodule update -f --init --recursive
Submodule path 'Tools/jMAVSim': checked out '3bd51e67e022ce59644d33ebdf6570c2ea88ddb6'
Submodule path 'Tools/jMAVSim/jMAVlib': checked out 'b8d4e8e7acfd2e47f4b51f0c2577fba6ef5ba735'
Submodule path 'Tools/sitl_gazebo': checked out '169d48217df89922e9fae72ef34fa46ce2e209dd'
Submodule path 'Tools/sitl_gazebo/external/OpticalFlow': checked out '54471159d4202d305e5643bbad3f2307f5dd1e37'
Submodule path 'Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker': checked out 'a0d242294ef638d8fa422ab43f0d476ba37a15a6'
Submodule path 'boards/atlflight/cmake_hexagon': checked out '08fd0a73045346448adf6969660196228b23e1fa'
Submodule path 'cmake/configs/uavcan_board_ident': checked out '2e5f9d6768b1dbffc006dc2ceeb2bfe120f22163'
Submodule path 'mavlink/include/mavlink/v2.0': checked out '086f176b677b719c1f5b4267a10553ffc5c3a974'
Submodule path 'msg/tools/gencpp': checked out '7e446a9976916a7b6fc7266098c67fc6f73a76e0'
Submodule path 'msg/tools/genmsg': checked out '5736b1f7ad037fb5811a3100ba9da2db0ec1f20a'
Submodule path 'platforms/nuttx/NuttX/apps': checked out '13eeff0ec3f23d84fabf1e9fe216846957ad15cf'
Submodule path 'platforms/nuttx/NuttX/nuttx': checked out '427238133be2b0ecd068a11e886ee8fdbc31f6dc'
Submodule path 'src/drivers/gps/devices': checked out '69e38ecc846df03f0bb71aa136bcc830dd3316a9'
Submodule path 'src/drivers/uavcan/libuavcan': checked out '6174b8c10a2dbf47076ca7a7b5820a4c36c8a988'
Submodule path 'src/drivers/uavcan/libuavcan/dsdl': checked out '192295c4f9b67f4a20b0eabf74757b6597415f2b'
Submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
Submodule path 'src/drivers/uavcan/libuavcan/libuavcan_drivers/kinetis': checked out '0c774a5a99bbd91d1f4832290fbed9168b2f65e5'
Submodule path 'src/lib/DriverFramework': checked out 'c7d2fe2d4f5954b9042af4edf7e5aeb21dbdd129'
Submodule path 'src/lib/DriverFramework/dspal': checked out '9b46b4a57f230672ee0806a523963af70bc44f1c'
Submodule path 'src/lib/DriverFramework/dspal/cmake_hexagon': checked out '07168bd5715818802b78f674816ec851307998a7'
Submodule path 'src/lib/ecl': checked out 'dec1ed0aeb395a885b60b1d61cfb57bd932c7d34'
Submodule path 'src/lib/matrix': checked out '92d1c8761ec8e5dbe9ba3f756cebd5e90956a1b2'
Submodule path 'src/modules/micrortps_bridge/micro-CDR': checked out '62d95c870eafb0cccc7bf70bb8a0cbb86f125a0e'
```