PX4/jMAVSim #124: New PX4 firmware + jMAVSim fails to build on Windows 10

copterspace, 2020-11-20:
Installed new Windows Toolchain 0.9 on Windows 10 Professional x64, then, as usial:
```bash
git clone --recursive -j8 https://github.com/PX4/Firmware.git
cd Firmware
make px4_sitl jmavsim

```
fails with compilation error:
```
$ make px4_sitl jmavsim
-- PX4 version: v1.11.0-rc3-727-g0db70ea475
-- PX4 config file: /cygdrive/c/PX4/home/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++.exe
-- Check for working CXX compiler: /usr/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3")
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-470-gda9f314
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
[0/761] git submodule src/drivers/gps/devices
[6/761] git submodule src/lib/ecl
[11/761] git submodule mavlink/include/mavlink/v2.0
[13/761] git submodule Tools/jMAVSim
[16/761] Generating ../romfs_files.tar
FAILED: romfs_files.tar
cd /cygdrive/c/PX4/home/Firmware/ROMFS/px4fmu_common && /usr/bin/cmake.exe -E tar cf /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/romfs_files.tar init.d/airframes/1000_rc_fw_easystar.hil init.d/airframes/1001_rc_quad_x.hil init.d/airframes/1002_standard_vtol.hil init.d/airframes/1100_rc_quad_x_sih.hil init.d/airframes/2100_standard_plane init.d/airframes/2105_maja init.d/airframes/2106_albatross init.d/airframes/2200_mini_talon init.d/airframes/2507_cloudship init.d/airframes/3000_generic_wing init.d/airframes/3030_io_camflyer init.d/airframes/3031_phantom init.d/airframes/3032_skywalker_x5 init.d/airframes/3033_wingwing init.d/airframes/3034_fx79 init.d/airframes/3035_viper init.d/airframes/3036_pigeon init.d/airframes/3037_parrot_disco_mod init.d/airframes/3100_tbs_caipirinha init.d/airframes/4001_quad_x init.d/airframes/4003_qavr5 init.d/airframes/4009_qav250 init.d/airframes/4010_dji_f330 init.d/airframes/4011_dji_f450 init.d/airframes/4014_s500 init.d/airframes/4015_holybro_s500 init.d/airframes/4016_holybro_px4vision init.d/airframes/4017_nxp_hovergames init.d/airframes/4020_hk_micro_pcb init.d/airframes/4030_3dr_solo init.d/airframes/4031_3dr_quad init.d/airframes/4040_reaper init.d/airframes/4041_beta75x init.d/airframes/4050_generic_250 init.d/airframes/4051_s250aq init.d/airframes/4052_holybro_qav250 init.d/airframes/4053_holybro_kopis2 init.d/airframes/4060_dji_matrice_100 init.d/airframes/4070_aerofc init.d/airframes/4071_ifo init.d/airframes/4072_draco init.d/airframes/4073_ifo-s init.d/airframes/4080_zmr250 init.d/airframes/4090_nanomind init.d/airframes/4100_tiltquadrotor init.d/airframes/4250_teal init.d/airframes/4500_clover4 init.d/airframes/4900_crazyflie init.d/airframes/5001_quad_+ init.d/airframes/6001_hexa_x init.d/airframes/6002_draco_r init.d/airframes/7001_hexa_+ init.d/airframes/8001_octo_x init.d/airframes/9001_octo_+ init.d/airframes/10015_tbs_discovery init.d/airframes/10016_3dr_iris init.d/airframes/10017_steadidrone_qu4d init.d/airframes/10018_tbs_endurance init.d/airframes/11001_hexa_cox init.d/airframes/12001_octo_cox init.d/airframes/12002_steadidrone_mavrik init.d/airframes/13000_generic_vtol_standard init.d/airframes/13001_caipirinha_vtol init.d/airframes/13002_firefly6 init.d/airframes/13003_quad_tailsitter init.d/airframes/13004_quad+_tailsitter init.d/airframes/13005_vtol_AAERT_quad init.d/airframes/13006_vtol_standard_delta init.d/airframes/13007_vtol_AAVVT_quad init.d/airframes/13008_QuadRanger init.d/airframes/13009_vtol_spt_ranger init.d/airframes/13010_claire init.d/airframes/13012_convergence init.d/airframes/13013_deltaquad init.d/airframes/13014_vtol_babyshark init.d/airframes/13050_generic_vtol_octo init.d/airframes/13200_generic_vtol_tailsitter init.d/airframes/14001_tri_y_yaw+ init.d/airframes/14002_tri_y_yaw- init.d/airframes/15001_coax_heli init.d/airframes/16001_helicopter init.d/airframes/17002_TF-AutoG2 init.d/airframes/24001_dodeca_cox init.d/airframes/50000_generic_ground_vehicle init.d/airframes/50001_axialracing_ax10 init.d/airframes/50002_traxxas_stampede_2wd init.d/airframes/50003_aion_robotics_r1_rover init.d/airframes/50004_nxpcup_car_dfrobot_gpx init.d/airframes/60000_uuv_generic init.d/airframes/60001_uuv_hippocampus init.d/airframes/60002_uuv_bluerov2_heavy init.d/rc.airship_apps init.d/rc.airship_defaults init.d/rc.boat_defaults init.d/rc.fw_apps init.d/rc.fw_defaults init.d/rc.interface init.d/rc.io init.d/rc.logging init.d/rc.mc_apps init.d/rc.mc_defaults init.d/rcS init.d/rc.sensors init.d/rc.thermal_cal init.d/rc.rover_apps init.d/rc.rover_defaults init.d/rc.uuv_apps init.d/rc.uuv_defaults init.d/rc.vehicle_setup init.d/rc.vtol_apps init.d/rc.vtol_defaults mixers/AAERTWF.main.mix mixers/AAVVTWFF.main.mix mixers/AAVVTWFF_vtail.main.mix mixers/AERT.main.mix mixers/AETRFG.main.mix mixers/babyshark.main.mix mixers/blade130.main.mix mixers/caipi.main.mix mixers/CCPM.main.mix mixers/claire.aux.mix mixers/claire.main.mix mixers/cloudship.main.mix mixers/coax.main.mix mixers/delta.main.mix mixers/deltaquad.main.mix mixers/dodeca_bottom_cox.aux.mix mixers/dodeca_top_cox.main.mix mixers/firefly6.aux.mix mixers/firefly6.main.mix mixers/fw_generic_wing.main.mix mixers/FX79.main.mix mixers/generic_diff_rover.main.mix mixers/hexa_cox.main.mix mixers/hexa_+.main.mix mixers/hexa_x.main.mix mixers/IO_pass.main.mix mixers/mount.aux.mix mixers/mount_legs.aux.mix mixers/octo_cox.main.mix mixers/octo_cox_w.main.mix mixers/octo_+.main.mix mixers/octo_x.main.mix mixers/pass.aux.mix mixers/phantom.main.mix mixers/quad_dc.main.mix mixers/quad_h.main.mix mixers/quad_+.main.mix mixers/quad_s250aq.main.mix mixers/quad_+_vtol.main.mix mixers/quad_w.main.mix mixers/quad_x_cw.main.mix mixers/quad_x.main.mix mixers/quad_x_vtol.main.mix mixers/rover_diff_and_servo.main.mix mixers/rover_generic.main.mix mixers/stampede.main.mix mixers/standard_vtol_hitl.main.mix mixers/TF-AutoG2.main.mix mixers/tilt_quad.aux.mix mixers/tilt_quad.main.mix mixers/tri_y_yaw+.main.mix mixers/tri_y_yaw-.main.mix mixers/uuv_x.main.mix mixers/vectored6dof.main.mix mixers/Viper.main.mix mixers/vtol_AAERT.aux.mix mixers/vtol_AAVVT.aux.mix mixers/vtol_convergence.main.mix mixers/vtol_delta.aux.mix mixers/vtol_tailsitter_duo.main.mix mixers/wingwing.main.mix mixers-sitl/autogyro_sitl.main.mix mixers-sitl/boat_sitl.main.mix mixers-sitl/delta_wing_sitl.main.mix mixers-sitl/plane_sitl.main.mix mixers-sitl/quad_x_vtol.main.mix mixers-sitl/rover_ackermann_sitl.main.mix mixers-sitl/rover_diff_sitl.main.mix mixers-sitl/standard_vtol_sitl.main.mix mixers-sitl/tiltrotor_sitl.main.mix mixers-sitl/uuv_x_sitl.main.mix mixers-sitl/vectored6dof_sitl.main.mix init.d-posix/airframes/10016_iris init.d-posix/airframes/10020_if750a init.d-posix/airframes/10030_px4vision init.d-posix/airframes/1010_iris_opt_flow init.d-posix/airframes/1010_iris_opt_flow.post init.d-posix/airframes/1011_iris_irlock init.d-posix/airframes/1012_iris_rplidar init.d-posix/airframes/1013_iris_vision init.d-posix/airframes/1013_iris_vision.post init.d-posix/airframes/1014_solo init.d-posix/airframes/1015_iris_obs_avoid init.d-posix/airframes/1015_iris_obs_avoid.post init.d-posix/airframes/1016_iris_rtps init.d-posix/airframes/1016_iris_rtps.post init.d-posix/airframes/1017_iris_opt_flow_mockup init.d-posix/airframes/1018_iris_vision_velocity init.d-posix/airframes/1019_iris_dual_gps init.d-posix/airframes/1020_uuv_generic init.d-posix/airframes/1021_uuv_hippocampus init.d-posix/airframes/1022_uuv_bluerov2_heavy init.d-posix/airframes/1030_plane init.d-posix/airframes/1031_plane_cam init.d-posix/airframes/1032_plane_catapult init.d-posix/airframes/1033_plane_lidar init.d-posix/airframes/1033_rascal init.d- [... cut ...] 6011_typhoon_h480.post init.d-posix/rc.replay init.d-posix/rcS
CMake Error: archive_write_header: archive_write_pax_header: 'x' header failed?!  This can't happen.

CMake Error: Problem creating tar: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/romfs_files.tar
[18/761] Generating git version header
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: px4_sitl] Error 1
```

Tried to checkout to previous releases (1.10, 1.09) - it did not help, same error.
How to repair it?...

copterspace, 2020-11-20:
Also tried to pull latest toolchain version 1.0 from github and run `install-all-components.bat`

This helped to repair build.

But now jmavsim does not start up - I just get this message:

```
$ make px4_sitl jmavsim
[0/1] cd /cygdrive/c/PX4/home/Firmware/build/px4_sitl_defa...mware /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe
debugger: none
program: jmavsim
model: none
world: none
src_path: /cygdrive/c/PX4/home/Firmware
build_path: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe" "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/cygdrive/c/PX4/home/Firmware"/test_data
INFO  [px4] Creating symlink /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/etc -> /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
/cygdrive/c/PX4/home/Firmware/Tools/jmavsim_run.sh: строка 59: ant: команда не найдена
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  EKF2_MULTI_IMU: curr: 0 -> new: 3
  SENS_IMU_MODE: curr: 1 -> new: 0
  EKF2_MULTI_MAG: curr: 0 -> new: 2
  SENS_MAG_MODE: curr: 1 -> new: 0
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
```

And simulator does not start.

Camry-hub, 2021-05-01:
> Also tried to pull latest toolchain version 1.0 from github and run `install-all-components.bat`
>
> This helped to repair build.
>
> But now jmavsim does not start up - I just get this message:
>
> ```
> $ [...]
У меня такая же проблема. Как починили не подскажете?

copterspace, 2021-05-06:
Перешёл на Ubuntu.