PX4/jMAVSim #133: java.lang.IllegalAccessException when running/building on Ubuntu 20.04

Manodiestra, 2021-11-23:
OS: Ubuntu 20.04
java: openjdk 16.0.1 2021-04-20
ant: 1.10.7

I get an error:
```
java.lang.IllegalAccessException: class javax.media.j3d.JoglPipeline cannot access class sun.awt.X11GraphicsDevice
```

I cloned the repo and tried building as specified in the README.md:
```
ant create_run_jar copy_res
cd out/production
java -Djava.ext.dirs= -jar jmavsim_run.jar
```
I also tried building the distribution package
```
ant distro
```
Then unzipped it and ran it from the command line with `java -jar jmavsim_run.jar`

In both cases it stops with error output as follows:
(Full output)
```
Options parsed, starting Sim.
Starting GUI...
Caught AppContextInfo(Bug 1004) InaccessibleObjectException: Unable to make public static sun.awt.AppContext sun.awt.AppContext.getAppContext() accessible: module java.desktop does not "exports sun.awt" to unnamed module @6c80d78a on thread J3D-Renderer-1
    [0]: java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
    [1]: java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
    [2]: java.base/java.lang.reflect.Method.checkCanSetAccessible(Method.java:199)
    [3]: java.base/java.lang.reflect.Method.setAccessible(Method.java:193)
    [4]: com.jogamp.nativewindow.awt.AppContextInfo$1$1.run(AppContextInfo.java:40)
    [5]: com.jogamp.common.util.UnsafeUtil.doWithoutIllegalAccessLogger(UnsafeUtil.java:202)
    [6]: com.jogamp.nativewindow.awt.AppContextInfo$1.run(AppContextInfo.java:34)
    [7]: java.base/java.security.AccessController.doPrivileged(AccessController.java:312)
    [8]: com.jogamp.nativewindow.awt.AppContextInfo.<clinit>(AppContextInfo.java:31)
    [9]: com.jogamp.nativewindow.awt.JAWTWindow.<init>(JAWTWindow.java:128)
    [10]: jogamp.nativewindow.jawt.x11.X11JAWTWindow.<init>(X11JAWTWindow.java:60)
    [11]: java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
    [12]: java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:78)
    [13]: java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
    [14]: java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
    [15]: java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
    [16]: jogamp.nativewindow.NativeWindowFactoryImpl.getAWTNativeWindow(NativeWindowFactoryImpl.java:105)
    [17]: jogamp.nativewindow.NativeWindowFactoryImpl.getNativeWindowImpl(NativeWindowFactoryImpl.java:66)
    [18]: com.jogamp.nativewindow.NativeWindowFactory.getNativeWindow(NativeWindowFactory.java:654)
    [19]: javax.media.j3d.JoglPipeline$QueryCanvas.addNotify(JoglPipeline.java:8604)
    [20]: java.desktop/java.awt.Container.addNotify(Container.java:2801)
    [21]: java.desktop/java.awt.Window.addNotify(Window.java:787)
    [22]: java.desktop/java.awt.Frame.addNotify(Frame.java:493)
    [23]: java.desktop/java.awt.Window.show(Window.java:1049)
    [24]: java.desktop/java.awt.Component.show(Component.java:1720)
    [25]: java.desktop/java.awt.Component.setVisible(Component.java:1667)
    [26]: java.desktop/java.awt.Window.setVisible(Window.java:1032)
    [27]: javax.media.j3d.JoglPipeline.getBestConfiguration(JoglPipeline.java:8379)
    [28]: javax.media.j3d.Renderer.doWork(Renderer.java:496)
    [29]: javax.media.j3d.J3dThread.run(J3dThread.java:271)
Caught AppContextInfo(Bug 1004) IllegalAccessException: class com.jogamp.nativewindow.awt.AppContextInfo cannot access class sun.awt.AppContext (in module java.desktop) because module java.desktop does not export sun.awt to unnamed module @6c80d78a on thread J3D-Renderer-1
    [0]: java.base/jdk.internal.reflect.Reflection.newIllegalAccessException(Reflection.java:385)
    [1]: java.base/java.lang.reflect.AccessibleObject.checkAccess(AccessibleObject.java:687)
    [2]: java.base/java.lang.reflect.Method.invoke(Method.java:559)
    [3]: com.jogamp.nativewindow.awt.AppContextInfo.fetchAppContext(AppContextInfo.java:191)
    [4]: com.jogamp.nativewindow.awt.AppContextInfo.update(AppContextInfo.java:135)
    [5]: com.jogamp.nativewindow.awt.AppContextInfo.<init>(AppContextInfo.java:50)
    [6]: com.jogamp.nativewindow.awt.JAWTWindow.<init>(JAWTWindow.java:128)
    [7]: jogamp.nativewindow.jawt.x11.X11JAWTWindow.<init>(X11JAWTWindow.java:60)
    [8]: java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
    [9]: java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:78)
    [10]: java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
    [11]: java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
    [12]: java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
    [13]: jogamp.nativewindow.NativeWindowFactoryImpl.getAWTNativeWindow(NativeWindowFactoryImpl.java:105)
    [14]: jogamp.nativewindow.NativeWindowFactoryImpl.getNativeWindowImpl(NativeWindowFactoryImpl.java:66)
    [15]: com.jogamp.nativewindow.NativeWindowFactory.getNativeWindow(NativeWindowFactory.java:654)
    [16]: javax.media.j3d.JoglPipeline$QueryCanvas.addNotify(JoglPipeline.java:8604)
    [17]: java.desktop/java.awt.Container.addNotify(Container.java:2801)
    [18]: java.desktop/java.awt.Window.addNotify(Window.java:787)
    [19]: java.desktop/java.awt.Frame.addNotify(Frame.java:493)
    [20]: java.desktop/java.awt.Window.show(Window.java:1049)
    [21]: java.desktop/java.awt.Component.show(Component.java:1720)
    [22]: java.desktop/java.awt.Component.setVisible(Component.java:1667)
    [23]: java.desktop/java.awt.Window.setVisible(Window.java:1032)
    [24]: javax.media.j3d.JoglPipeline.getBestConfiguration(JoglPipeline.java:8379)
    [25]: javax.media.j3d.Renderer.doWork(Renderer.java:496)
    [26]: javax.media.j3d.J3dThread.run(J3dThread.java:271)
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:78)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:567)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.RuntimeException: java.lang.IllegalAccessException: class javax.media.j3d.JoglPipeline cannot access class sun.awt.X11GraphicsDevice (in module java.desktop) because module java.desktop does not export sun.awt to unnamed module @6c80d78a
	at javax.media.j3d.JoglPipeline.getScreen(JoglPipeline.java:8553)
	at javax.media.j3d.Screen3D.<init>(Screen3D.java:354)
	at javax.media.j3d.Canvas3D.<init>(Canvas3D.java:1124)
	at javax.media.j3d.Canvas3D.<init>(Canvas3D.java:1026)
	at javax.media.j3d.Canvas3D.<init>(Canvas3D.java:990)
	at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.<init>(Visualizer3D.java:909)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:196)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:192)
	at me.drton.jmavsim.Simulator.main(Simulator.java:941)
	... 5 more
Caused by: java.lang.IllegalAccessException: class javax.media.j3d.JoglPipeline cannot access class sun.awt.X11GraphicsDevice (in module java.desktop) because module java.desktop does not export sun.awt to unnamed module @6c80d78a
	at java.base/jdk.internal.reflect.Reflection.newIllegalAccessException(Reflection.java:385)
	at java.base/java.lang.reflect.AccessibleObject.checkAccess(AccessibleObject.java:687)
	at java.base/java.lang.reflect.Method.invoke(Method.java:559)
	at javax.media.j3d.JoglPipeline.getScreen(JoglPipeline.java:8551)
	... 13 more
```

Any help to resolve the issue would be appreciated.

Manodiestra, 2022-01-04:
The issue was the use of OpenJDK. The project requires a proprietary JDK to build.

closed 2022-01-04 by Manodiestra

jzy3d, 2022-01-10:
You may also keep your existing JDK 16 and use JVM option `--illegal-access=permit`

https://github.com/jzy3d/jogl/issues/11

alper50, 2022-02-14:
can someone help me ı am getting same error but ı couldnt understand the answers, detailed answer could be useful

Manodiestra, 2022-02-17:
@alper50 
In your terminal run `java --version` and see what it spits out. If it's pulling OpenJDK, then either uninstall it or switch the default Java JDK in your shell PATH.

Or you can try doing what @jzy3d suggested and run `java --illegal-access=permit -jar my.jar` instead of `java -jar my.jar`.

alper50, 2022-03-19:
java --version output is = openjdk 17.0.1 2021-10-19
OpenJDK Runtime Environment (build 17.0.1+12)
OpenJDK 64-Bit Server VM (build 17.0.1+12, mixed mode)

and ı decided to run java --illegal-access=permit -jar my.jar but the output is = ' Error: Unable to access jarfile my.jar '