PX4/jMAVSim #134: Proposal: Replacing ExecutorService to achieve more stable timing in the update loop

ecmnet, 2022-07-05:
The update loop is currently scheduled by an ExecutorService, which is known to be not very precise in its cyclic timing.

Therefore I would like to propose another solution for scheduling, which I currently use to avoid or at least reduce issues like polling errors or timeouts e.g. mentioned in this [discussion](https://discuss.px4.io/t/px4-poll-timeout/20484).

Scheduling is implemented like this:

        long sleep_interval_ns;
        if (LOCKSTEP_ENABLED) {`
        	sleep_interval_ns = (long)(sleepInterval / speedFactor / checkFactor)*1000;
        } else {
        	sleep_interval_ns = sleepInterval*1000;
        }
        Thread loop = new Thread(() -> {
    		long  wait;
    		while(!shutdown) {
    			wait = System.nanoTime();
    			this.run();
    			LockSupport.parkNanos(sleep_interval_ns - (System.nanoTime() - wait));    
    		}
    	});
        loop.setPriority(Thread.MAX_PRIORITY);
        loop.start();

On OSX this turned out to be much more stable although it is not solving the problem completely.

dagar, 2022-07-05:
@ecmnet what actual requirements do you have for simulation? I'm wondering if it would be better to focus on using `sih` within PX4 (https://github.com/PX4/PX4-Autopilot/tree/main/src/modules/sih) for the this simple sim, then you can continue using jmavsim only for visualization.

For anything more sophisticated I'm hoping we can focus our efforts on new Gazebo (Ignition), including dealing with cross platform build (or packaging) issues if necessary.

ecmnet, 2022-07-05:
@dagar: I use jMAVSim just for quick and basic functional tests of [MAVCGL](https://github.com/ecmnet/MAVGCL) or the companion software (e.g. trajectory generation w. offboard) - usually without visualization. For more sophisticated tests, I currently use Gazebo but hadn't have a look into Ignition.

dagar, 2022-07-05:
Makes sense. In the not too distant future I would like to consolidate a bit so we can focus on having 1 or 2 decent simulation options instead of 5-6 mediocre.